pasqal-io/Pulser #726: Add option to draw layout associated to register on the same plot as the Register

a-corni, 2024-08-28:
It is currently hard to visualize what trapping spots have been selected among a layout to define a Register. The layout is stored inside a register. With the option "qubits_color", we can implement the plotting of both layout and register on the same plot:
[image]

BrFle, 2024-10-02:
Hey ! I'd like to start contributing on this project, can I be assigned this issue ?

HGSilveri, 2024-10-02:
Hey @BrFle , of course! Let us know if you need any help :)

a-corni, 2024-12-02:
Hi @BrFle, do you need help to tackle this issue ? 😄

BrFle, 2024-12-03:
> Hi @BrFle, do you need help to tackle this issue ? 😄
Hey @a-corni ! I got confused and never published my PR. I got something working, i'm still lacking reasonable tests. Can you go take a look ?
Thanks a lot !
https://github.com/pasqal-io/Pulser/pull/772

a-corni, 2024-12-03:
Thanks @BrFle ! I will have a look !

PR #772 (2025-05-16): Draw layout and register on the same plot
First draft to add option to draw the layout.
Still missing tests !
Fixes #726
PR #865 (2025-05-19): Draw layout and register on the same plot
Add an option to draw the empty traps in a register.
Fixes #726 
Continues #772
closed 2025-05-19 by a-corni