pasqal-io/Pulser #730: Upgrade to Qutip 5

HGSilveri, 2024-09-10:


HGSilveri, 2024-09-23:
I gave this a shot and the conclusions were underwhelming (to say the least):
- `Qobj` is not longer `ArrayLike`, so we have to explicitly call `full()` to use numpy ufuncs and sometimes go back to `Qobj`
- Some methods (eg `Qobj.extract_states()`, `Qobj.eliminate_states()`) were deleted without a substitute, so we have to implement them manually
- **Execution became dramatically slower**: On my laptop, `test_simulation.py` ran in 8.55s on `qutip==4.7.6`, now doesn't even complete in _10 minutes!_ (I gave up and interrupted it)

It's clear that Qutip 5 introduced changes that are incompatible with our current implementation in `pulser-simulation`. Until that changes, we should stick with Qutip 4.

awennersteen, 2024-10-23:
Re: performance there appears to have been a few other reports of this that they were able to fix, but in some cases they're pretty low level

https://github.com/qutip/qutip/issues/2443
https://github.com/qutip/qutip/issues/2406

However, given #757 we might not be able to postpone this forever.

HGSilveri, 2024-10-23:
Thanks for looking into this @awennersteen . So far, these changes have not been released yet so I can't tell whether they will help without getting into the weeds of installing qutip from source. As this is not urgent, I'll wait for the release before trying again.

PR #783 (2024-12-18): Upgrade pulser-simulation to qutip 5
Bumps the requirements to qutip 5.
Handles the introduction of the new data type "Dense" in qutip (previously, there was only "CSR"): by default, all the operators used to be initialized as "CSR", but they were now stored as "Dense". This PR converts any Qobj to CSR, such that the performances are kept the same (notably, having CSR in collapse operators is very important, see https://github.com/qutip/qutip/issues/2328). 
- Here is the benchmark performed with qutip 4.7.6: [benchmark_qutip_4_7_6.pdf](https://github.com/user-attachments/files/18154270/benchma [... cut]
closed 2024-12-18 by a-corni