pasqal-io/Pulser #742: UserWarning: 'runs' is not used by any active noise type ('SPAM',) when measurement errors without state preparation errors

pablolh, 2024-10-01:
Dear team,

It seems Pulser requires `runs` and `samples_per_run` to NOT be specified in `NoiseModel` when
`p_false_pos > 0` and `state_prep_error == 0`.

To reproduce, execute the following script
```
import pulser

noise_model = pulser.noise_model.NoiseModel(
            runs=1,
            samples_per_run=1,
            p_false_pos=0.1,
        )
```

Output: 
```
UserWarning: 'runs' is not used by any active noise type ('SPAM',).
  noise_model = pulser.noise_model.NoiseModel(
UserWarning: 'samples_per_run' is not used by any active noise type ('SPAM',).
  noise_model = pulser.noise_model.NoiseModel(
```

However, as soon as `state_prep_error` is `>0` then the warning disappears.
This is unexpected, since then the active noise types are then exactly the same, so the warning has an issue, I believe.
I would expect no warning in either case, or dissociate state preparation (which would require `runs`) from measurement errors (which maybe wouldn't). Let me know if that makes sense.

Using `pulser-core` 0.20.1

HGSilveri, 2024-10-02:
Hey @pablolh , thanks for raising this issue.

> However, as soon as `state_prep_error` is `>0` then the warning disappears.
> This is unexpected, since then the active noise types are then exactly the same, so the warning has an issue, I believe.

The fact that SPAM encompasses both state preparation errors (which require `runs` and `samples_per_run` due to how they are emulated) and measurement errors (which are all added in post-processing, so they don't require these parameters) is indeed the source of confusion here. 

> It seems Pulser requires `runs` and `samples_per_run` to NOT be specified in `NoiseModel` when `p_false_pos > 0` and `state_prep_error == 0`.

I would say this is a bit too strong of a statement: Pulser is not _requiring_ `runs` and `samples_per_run` to be not be specified when `state_prep_error==0`, it is only warning the user that these parameters (which the user purposefully specified) won't be used in this case. 

> I would expect no warning in either case, or dissociate state preparation (which would require `runs`) from measurement errors (which maybe wouldn't). Let me know if that makes sense. 

The noise types are no longer explicitly provided by the user (ie the user is not setting `noise_types=("SPAM",)`) so I would argue they are already dissociated. If you care only about measurement errors (or even just false positives), you'd give
```
noise_model = pulser.noise_model.NoiseModel(
            p_false_pos=0.1,
        )
```
while if you want only state preparation errors, you'd specify eg
```
noise_model = pulser.noise_model.NoiseModel(
            state_prep_error=0.1,
            runs=10,
            samples_per_run=1,
        )
```
Nonetheless, I agree that the warning message just mentioning "SPAM" is not clear, so perhaps adding which noise parameters are "active" would help with that. Is this what you had in mind?

pablolh, 2024-10-03:
Hi Henrique,
Thanks for the quick answer!
I agree with what you say and I got indeed only mislead by the warning message itself.
If a different setup is required for state preparation and measurement errors then only mentioning SPAM in the warning is a bit confusing.
The functionality itself has no problem.

closed 2024-10-16 by HGSilveri