pasqal-io/Pulser #756: Show all the specs of the `Device`

a-corni, 2024-10-21:
Current `_specs` of the Device is only implemented for part of the attributes of the Device. For instance it does not display `default_noise_model`, `requires_layout`, `accepts_new_layout`, ...

HaroldErbin, 2024-11-19:
I had a few questions related to this point:
1. Why not store also the properties in a `dict`, in case we want to pass all properties (for example, to define a new device, save as json for easy check of the specs, etc.)? I find having a string not always convenient (also it's supposed to be an internal method).
2. Why not implement `_specs` also for virtual devices? Following [this tutorial](https://pulser.readthedocs.io/en/stable/tutorials/virtual_devices.html), we could expect to be able to print the specs also for the derived virtual device.
3. Why implement `_specs` as an internal method and not as a public property?

HaroldErbin, 2024-11-19:
Note that I just found the `_params()` internal method, which does work for both physical and virtual devices. However, it does not contain all parameters (for example, `interaction_coefficient` is missing, while `interaction_coefficient_xy` is present). It could also better be a public property.

closed 2024-12-16 by HGSilveri