pasqal-io/Pulser #757: Pulser doesn't install on Python 3.13

awennersteen, 2024-10-22:
Trying to install Pulser on Python 3.13 fails due to issues with building QuTiP and also the supported SciPy version of QuTiP. 

Here is a way to replicate via Docker containers, the same process works with Python 3.12 container (you don't need to install anything via apt with python 3.12 as everything comes pre-built)

The issue appears to come from restricting SciPy < 1.13 in pulser-simulation:
https://github.com/pasqal-io/Pulser/pull/668
I checked and this error still persists in SciPy 1.14.1,  and is marked as unsupported also in QuTiP now from 4.7.6.

The QuTiP  issue of course is that we haven't upgraded to version 5, as detailed in #730 .

```bash
docker pull python:3.13
docker run -it python bash
root@bc1448bacafb:/# apt update && apt install -y build-essential gfortran libopenblas-dev
root@bc1448bacafb:/# pip install pulser
```

Also trying to build qutip, bypassing scipy restriction, seems to fail independently:

```bash
      qutip/control/cy_grape.cpp: In function ‘int __Pyx_PyInt_As_int(PyObject*)’:
      qutip/control/cy_grape.cpp:7770:46: error: too few arguments to function ‘int _PyLong_AsByteArray(PyLongObject*, unsigned char*, size_t, int, int, int)’
       7770 |                 int ret = _PyLong_AsByteArray((PyLongObject *)v,
            |                           ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
       7771 |                                               bytes, sizeof(val),
            |                                               ~~~~~~~~~~~~~~~~~~~
       7772 |                                               is_little, !is_unsigned);
            |                                               ~~~~~~~~~~~~~~~~~~~~~~~~
      In file included from /usr/local/include/python3.13/longobject.h:107,
                       from /usr/local/include/python3.13/Python.h:81,
                       from qutip/control/cy_grape.cpp:34:
      /usr/local/include/python3.13/cpython/longobject.h:111:17: note: declared here
        111 | PyAPI_FUNC(int) _PyLong_AsByteArray(PyLongObject* v,
            |                 ^~~~~~~~~~~~~~~~~~~
````

HGSilveri, 2024-12-20:
It seems we jumped the gun on this one, `pyqtorch` still does not support Python 3.13 on MacOS and Windows. 
Let's track the progress here: https://github.com/pytorch/pytorch/issues/130249

HGSilveri, 2025-03-26:
I tried to get back to this. It seems that from `torch>=2.6`, Python 3.13 is supported so that's no longer an issue.
However, I still have a blocker:
- It seems that there is no built wheel of `numpy 1` for Python 3.13, so when installing `pip` downloads the source and builds the wheel itself
- This seems to be working abnormally on Windows, because I then get this error when `matplotlib` attempts to import `numpy`
```
 ImportError while loading conftest 'D:\a\Pulser\Pulser\tests\conftest.py'.
tests\conftest.py:19: in <module>
    import matplotlib.pyplot as plt
C:\hostedtoolcache\windows\Python\3.13.2\x64\Lib\site-packages\matplotlib\__init__.py:161: in <module>
    from . import _api, _version, cbook, _docstring, rcsetup
C:\hostedtoolcache\windows\Python\3.13.2\x64\Lib\site-packages\matplotlib\cbook.py:24: in <module>
    import numpy as np
C:\hostedtoolcache\windows\Python\3.13.2\x64\Lib\site-packages\numpy\__init__.py:135: in <module>
    raise ImportError(msg) from e
E   ImportError: Error importing numpy: you should not try to import numpy from
E           its source directory; please exit the numpy source tree, and relaunch
E           your python interpreter from there.
```
Not sure there is a workaround for this. Perhaps we can just not support the combo `python3.13 + windows + numpy1` once we support numpy 2.

pimvenderbosch, 2025-03-26:
> I tried to get back to this. It seems that from `torch>=2.6`, Python 3.13 is supported so that's no longer an issue. However, I still have a blocker:
> 
> ...
> 
> Not sure there is a workaround for this. Perhaps we can just not support the combo `python3.13 + windows + numpy1` once we support numpy 2.

I would say that it is acceptable to not support certain combos because of dependency conflicts. As long as the vast majority of combinations is supported and there is an upgrade option, in this case use numpy 2, available. Not sure if and how to document these cases, though.

PR #786 (2024-12-20): Support Python 3.13
Fixes #757 .
PR #840 (2025-03-26): Support Python 3.13
- Adds support for Python 3.13

Closes #757 .
closed 2025-03-26 by HGSilveri