pasqal-io/Pulser #761: Function to estimate delay needed between two pulses

HaroldErbin, 2024-11-18:
I was wondering if it would be possible to have a function in `Sequence` (or `Channel` or anywhere convenient) to estimate the delay needed between two pulses on a physical device?

Context: I want to simulate the dynamics for a global pulse for the maximum duration allowed by `AnalogDevice`. In some cases, I want to start the dynamics with a "short" pulse to change the initial state. To stay below `device.max_sequence_duration`, I want to reduce the duration of the long pulse. This is easy for `MockDevice` since the pulses are sharp, but the pulses are modified for `AnalogDevice` to respect the physical constraints of the device. In particular, a delay can be added between the two pulses. Since I cannot get the duration of the delay before adding the long pulse to the sequence, but `seq.add()` raises an error because maximal duration is exceeded, I am stuck:
```
RuntimeError: The sequence's duration exceeded the maximum duration allowed by the device (4000 ns).
```

Here is the relevant part of my code:
```python
device = pulser.AnalogDevice
channel = seq.declared_channels["ising"]
seq = pulser.Sequence(register, device)
seq.declare_channel("ising", "rydberg_global")

pi2_wf = pulser.BlackmanWaveform.from_max_val(channel.max_amp, np.pi / 2)
hadamard = pulser.Pulse.ConstantDetuning(
    amplitude=pi2_wf, detuning=0, phase=np.pi / 2, post_phase_shift=np.pi
)
seq.add(hadamard, "ising")

duration = 4000
cst_pulse = pulser.Pulse.ConstantPulse(
        duration=duration, amplitude=Omega, detuning=delta, phase=0.
)
# this add delay but also raises error
seq.add(hadamard, "ising")
```

I tried to investigate the source, but did not manage to extract a simple function to get the delay. For example, I thought that this would work:
```python
state_prep_duration = hadamard.duration
delay = seq._schedule._find_add_delay(state_prep_duration, channel="ising", protocol="min-delay")
```
but it does not give the delay observed inside the full sequence (which I get by just decreasing duration until it works, which is not generalizable):
```
Channel: ising
t: 0 | Initial targets: 0, 1, 2, 3 | Phase Reference: 0.0 
t: 0->500 | Pulse(Amp=Blackman(Area: 1.57) rad/µs, Detuning=0 rad/µs, Phase=1.57) | Targets: 0, 1, 2, 3
t: 500->696 | Delay 
t: 696->3956 | Pulse(Amp=5 rad/µs, Detuning=5 rad/µs, Phase=3.14) | Targets: 0, 1, 2, 3
```

HaroldErbin, 2024-11-19:
Hi @HGSilveri . Thanks a lot for your reply!
After posting, I had arrived at the following:
```python
full_delay = hadamard.get_full_duration(ising_ch_obj) - hadamard.duration + ising_ch_obj.phase_jump_time
```
but I was not sure why it worked, and your explanations are clarifying it.

HGSilveri, 2024-11-19:
> Hi @HGSilveri . Thanks a lot for your reply! After posting, I had arrived at the following:
> 
> ```python
> full_delay = hadamard.get_full_duration(ising_ch_obj) - hadamard.duration + ising_ch_obj.phase_jump_time
> ```
> 
> but I was not sure why it worked, and your explanations are clarifying it.

Indeed, this is equivalent to what I suggested and should give exactly the same result :)

PR #773 (2024-12-04): Add method to Sequence to estimate delay introduced before adding pulse
Adds the method `estimate_add_delay` to Sequence.
Has the same signature as the `add` method.
Takes into account AOM and EOM Pulses (raise a Warning if the Pulse added is not Constant, with amplitude and detuning equal to the rabi_fres and detuning_on of the EOM Block)
Raise a Warning if the duration of the added Pulse exceeds the duration of the Sequence.
Fixes #761
closed 2024-12-04 by a-corni