pasqal-io/Pulser #775: New release request: Flexibility on qutip + scipy extensions (Forschungszentrum Jülich/JSC)

AlvaroHuanay, 2024-12-05:
Dear Pulser team,

We are contacting you from Forschungszentrum Jülich. Developing [Pulser 1.1.1](https://github.com/pasqal-io/Pulser/tree/v1.1.1) for our Stage 2025, we came accross with an incompatibility issue. I would like to ask if it would be possible to make a new Pulser release compatible with the following libraries:

qutip==5.0.4
scipy==1.13.1

The other dependencies you have for Pulser fit perfectly for us. Those are the only extensions we have issues with. The requirements you have in your last release for these extensions are:

[qutip~=4.7.5](https://github.com/pasqal-io/Pulser/blob/3c76eddcc2da7026518917ecea341cc9ffe389dc/pulser-simulation/requirements.txt#L1)
[scipy<1.13](https://github.com/pasqal-io/Pulser/blob/3c76eddcc2da7026518917ecea341cc9ffe389dc/pulser-simulation/requirements.txt#L3C1-L3C11)

Could you release a new version with these compatibilities?

Any additional information you may need I will be glad to provide you so.

Kind regards.

HGSilveri, 2024-12-05:
Hello @AlvaroHuanay , thanks for reaching out!

I'm afraid this will not be straigthforward because we encountered a severe performance degradation when we first tried to upgrade to Qutip 5 (see #730 for details). There has been no Qutip release since we last tried so I don't expect things to have changed on that front.

If I may ask, do you happen to know if you actually need to run emulations on this environment with restricted dependencies? I ask because these conflicts come from the `pulser-simulation` package, which is only needed for emulations. All sequence creation functionality is in `pulser-core`, so you could eventually just install `pulser-core` instead of `pulser` - which includes both packages.

AlvaroHuanay, 2024-12-05:
Hi @HGSilveri 

Thank you very much for your quick update. We are trying to get Qutip 4.7.5 on our systems. Nonetheless, we would really appreciate if the extension "scipy==1.13.1" could be added at least to be able to integrate it.

Our intention is to have the complete pulser module as we will use both tools. Do you have in mind upgrading to Qutip 5.X for later releases? I ask this based on the coments on https://github.com/pasqal-io/Pulser/issues/730, https://github.com/pasqal-io/Pulser/issues/757.

Kind regards.

awennersteen, 2024-12-05:
Hi @AlvaroHuanay,

Unfortunately that wouldn't help you much as QuTiP is also incompatible with Scipy>=1.13.
Hence that would involve patching QuTiP as well. I know and understand the JSCs desire to avoid another older SciPy install, but in this case in order to ensure the stability and quality of the software I would ask that you consider it. 

QuTiP PR for the restriction on their side:
https://github.com/qutip/qutip/pull/2383

As mentioned in the issues you linked we will have to resolve this, but I don't think we can commit to a timeline immediately. Henrique will know better.

Best,
Aleksander

HGSilveri, 2024-12-05:
Hi again @AlvaroHuanay 

> Thank you very much for your quick update. We are trying to get Qutip 4.7.5 on our systems. Nonetheless, we would really appreciate if the extension "scipy==1.13.1" could be added at least to be able to integrate it.

I'm afraid this won't be possible because the Qutip 4.7.6 (the latest and probably last release of Qutip 4) enforces `scipy<1.13`; there is some issue between Qutip 4 and `scipy>=1.13`, so they enforced this restriction. It is the only reason we have it as well, actually (as @awennersteen just pointin out :sweat_smile: ).

> Our intention is to have the complete pulser module as we will use both tools. Do you have in mind upgrading to Qutip 5.X for later releases? I ask this based on the coments on #730, #757.

We will take another crack at this very soon. We can keep you up to date on how it goes but it is indeed not possible to commit to a timeline at the moment.

AlvaroHuanay, 2024-12-06:
Greetings,

Thank you both (@awennersteen and @HGSilveri) for the explanation. I contacted our team and we are going to use pulser-core in the meantime. Regarding @HGSilveri last comment:

> We will take another crack at this very soon. We can keep you up to date on how it goes but it is indeed not possible to commit to a timeline at the moment.

 could you please reach to us when you are able to solve this issue with Qutip 5.X and Scipy 1.13.1? I will stay tunned on your repository as well as Qutip's repository to be aware of it just in case because our intention is being able to use the complete pulser module in our systems for the future.

Thank you very much, your comments were very helpful.

Kind Regards.

HGSilveri, 2024-12-20:
Hi @AlvaroHuanay ! 
We just released Pulser v1.2.0, which upgrades `pulser-simulation` to Qutip 5 and drops the Scipy restriction.
In principle it should solve your issue but feel free to reopen this if that's not the case.
Happy holidays!

closed 2024-12-20 by HGSilveri