pasqal-io/Pulser #789: Type of initial state is too strict

HaroldErbin, 2025-01-10:
It seems that the data type of `initial_state` in `pulser.EmulatorConfig` is too strict. Indeed, I am defining the state as a numpy array, and it is not recognized as a valid `Sequence[Complex]`.

Here is my code:
```python
if initial_state == "plus":
    initial_state = np.ones(2**L, dtype=np.complex128)
elif initial_state == "down":
    initial_state = "all-ground"
else:
    raise ValueError(f"Initial state '{initial_state}' is not valid.")

config = pulser.EmulatorConfig(
        sampling_rate=sampling_rate,
        evaluation_times="Full",
        initial_state=initial_state,
    )
```

PyLance in VSCode complains:
```
Argument of type "NDArray[float64] | Literal['all-ground']" cannot be assigned to parameter "initial_state" of type "Sequence[complex] | Literal['all-ground']" in function "__init__"
  Type "NDArray[float64] | Literal['all-ground']" is not assignable to type "Sequence[complex] | Literal['all-ground']"
    Type "NDArray[float64]" is not assignable to type "Sequence[complex] | Literal['all-ground']"
      "ndarray[Any, dtype[float64]]" is not assignable to "Sequence[complex]"
      "ndarray[Any, dtype[float64]]" is not assignable to "Literal['all-ground']"Pylance[reportArgumentType](https://github.com/microsoft/pyright/blob/main/docs/configuration.md#reportArgumentType)
(parameter) initial_state: NDArray[float64] | Literal['all-ground']
```
Specifying that `dtype=complex` gives `NDArray[Any]` and `dtype=np.complex128` gives `NDArray[complexfloating[_64Bit, _64Bit]]`, none of them recognized as `Sequence[complex]`.

a-corni, 2025-01-21:
Hello @HaroldErbin 
Thanks for pointing this out. Indeed, it seems that ndarrays are not detected as Sequence.
I will fix the typing

HGSilveri, 2025-02-04:
Closed by #792

PR #792 (2025-01-23): Fix typing of initial state in EmulatorConfig, fix normalization with qutip 5
Fixes #789  and #791
closed 2025-02-04 by HGSilveri