pasqal-io/Pulser #791: Bug with initial state in Pulser 1.2

HaroldErbin, 2025-01-20:
Specifying an initial state in Pulser 1.2 gives an erroneous behavior during the simulation.

Here is the configuration of the simulation:
```python
L = len(sequence.register.qubits)

if initial_state == "plus":
    initial_state = np.ones(2**L)
elif initial_state == "down":
    initial_state = "all-ground"
else:
    raise ValueError(f"Initial state '{initial_state}' is not valid.")

config = pulser.EmulatorConfig(
    sampling_rate=sampling_rate,
    evaluation_times="Full",
    initial_state=initial_state,
)
```

With `pulser==1.1.1`, here is what I get for `initial_state == "plus"` :

[image]

While for `pulser==1.2` I get:

[image]

Note that the curves are the same but the scale of y-axis different. My guess is that there is an issue with the normalization of the state, which could arise with some different convention between Qutip 4 and 5.
Normalizing the vector above does not help.

a-corni, 2025-01-21:
Hello @HaroldErbin,
Thanks for reaching out ! I think it's related to an option we have added when we moved to qutip 5, motivated by [this PR](https://github.com/qutip/qutip/issues/2406), we've set normalize_output=False` for the simulation to be always as fast: https://github.com/pasqal-io/Pulser/blob/develop/pulser-simulation/pulser_simulation/simulation.py#L583

a-corni, 2025-01-21:
For me to test it, could you send me the Sequence you are doing ? (you can send me the output of `Sequence.to_abstract_repr()`)

HaroldErbin, 2025-01-21:
Here is the output:
```json
{
    "version": "1",
    "name": "pulser-exported",
    "register": [
        {"name": "0", "x": -9.0, "y": 0.0},
        {"name": "1", "x": -3.0, "y": 0.0},
        {"name": "2", "x": 3.0, "y": 0.0},
        {"name": "3", "x": 9.0, "y": 0.0},
    ],
    "channels": {"ising": "rydberg_global"},
    "variables": {},
    "operations": [
        {
            "op": "pulse",
            "channel": "ising",
            "protocol": "min-delay",
            "amplitude": {
                "kind": "constant",
                "duration": 4000,
                "value": 9.277724408436214,
            },
            "detuning": {
                "kind": "constant",
                "duration": 4000,
                "value": 18.700413260754242,
            },
            "phase": 0.0,
            "post_phase_shift": 0.0,
        }
    ],
    "measurement": null,
    "device": {
        "name": "AnalogDevice",
        "dimensions": 2,
        "rydberg_level": 60,
        "min_atom_distance": 5,
        "max_atom_num": 100,
        "max_radial_distance": 35,
        "interaction_coeff_xy": null,
        "supports_slm_mask": false,
        "max_layout_filling": 0.5,
        "max_sequence_duration": 4000,
        "max_runs": 2000,
        "reusable_channels": false,
        "pre_calibrated_layouts": [
            {
                "coordinates": [
                    [-20.0, 0.0],
                    [-17.5, -4.330127],
                    [-17.5, 4.330127],
                    [-15.0, -8.660254],
                    [-15.0, 0.0],
                    [-15.0, 8.660254],
                    [-12.5, -12.990381],
                    [-12.5, -4.330127],
                    [-12.5, 4.330127],
                    [-12.5, 12.990381],
                    [-10.0, -17.320508],
                    [-10.0, -8.660254],
                    [-10.0, 0.0],
                    [-10.0, 8.660254],
                    [-10.0, 17.320508],
                    [-7.5, -12.990381],
                    [-7.5, -4.330127],
                    [-7.5, 4.330127],
                    [-7.5, 12.990381],
                    [-5.0, -17.320508],
                    [-5.0, -8.660254],
                    [-5.0, 0.0],
                    [-5.0, 8.660254],
                    [-5.0, 17.320508],
                    [-2.5, -12.990381],
                    [-2.5, -4.330127],
                    [-2.5, 4.330127],
                    [-2.5, 12.990381],
                    [0.0, -17.320508],
                    [0.0, -8.660254],
                    [0.0, 0.0],
                    [0.0, 8.660254],
                    [0.0, 17.320508],
                    [2.5, -12.990381],
                    [2.5, -4.330127],
                    [2.5, 4.330127],
                    [2.5, 12.990381],
                    [5.0, -17.320508],
                    [5.0, -8.660254],
                    [5.0, 0.0],
                    [5.0, 8.660254],
                    [5.0, 17.320508],
                    [7.5, -12.990381],
                    [7.5, -4.330127],
                    [7.5, 4.330127],
                    [7.5, 12.990381],
                    [10.0, -17.320508],
                    [10.0, -8.660254],
                    [10.0, 0.0],
                    [10.0, 8.660254],
                    [10.0, 17.320508],
                    [12.5, -12.990381],
                    [12.5, -4.330127],
                    [12.5, 4.330127],
                    [12.5, 12.990381],
                    [15.0, -8.660254],
                    [15.0, 0.0],
                    [15.0, 8.660254],
                    [17.5, -4.330127],
                    [17.5, 4.330127],
                    [20.0, 0.0],
                ],
                "slug": "TriangularLatticeLayout(61, 5.0\u00b5m)",
            }
        ],
        "accepts_new_layouts": false,
        "version": "1",
        "channels": [
            {
                "id": "rydberg_global",
                "basis": "ground-rydberg",
                "addressing": "Global",
                "max_abs_detuning": 125.66370614359172,
                "max_amp": 12.566370614359172,
                "min_retarget_interval": null,
                "fixed_retarget_t": null,
                "max_targets": null,
                "clock_period": 4,
                "min_duration": 16,
                "max_duration": 100000000,
                "mod_bandwidth": 8,
                "eom_config": {
                    "limiting_beam": "RED",
                    "max_limiting_amp": 188.49555921538757,
                    "intermediate_detuning": 2827.4333882308138,
                    "controlled_beams": ["BLUE"],
                    "mod_bandwidth": 40,
                    "custom_buffer_time": 240,
                },
            }
        ],
        "is_virtual": false,
    },
    "layout": {
        "coordinates": [
            [-12.681159, -3.550725],
            [-12.681159, 3.550725],
            [-9.0, 0.0],
            [-3.0, 0.0],
            [3.0, 0.0],
            [9.0, 0.0],
            [12.681159, -3.550725],
            [12.681159, 3.550725],
        ],
        "slug": "LineLayout({L}, {R_atom:.1}\u00b5m)",
    },
}
```

But the simulation is pretty simple: 3 atoms on a line spaced by 6 µm, and a sequence with constant amplitude and detuning:
```
Channel: ising
t: 0 | Initial targets: 0, 1, 2, 3 | Phase Reference: 0.0 
t: 0->4000 | Pulse(Amp=9.28 rad/µs, Detuning=18.7 rad/µs, Phase=0) | Targets: 0, 1, 2, 3
```

HaroldErbin, 2025-01-23:
Indeed, I had spotted the 8 = 2**3, but thought that normalizing the vector above did not work (I think I had made a mistake when dividing by the norm, which explains why I had not seen a change).

HGSilveri, 2025-01-24:
@HaroldErbin We released the fix in `v1.2.1` yesterday, the `initial_state` is now automatically normalized again, as before.

closed 2025-01-24 by HGSilveri