pasqal-io/Pulser #810: Add allclose() and transpose() to `pulser.math`

HGSilveri, 2025-02-13:
I suggest to have
- an allclose function in the init of pulser_math, since there is also torch.allclose https://pytorch.org/docs/main/generated/torch.allclose.html and it can useful for testing
- a transpose method in AbstractArray, since torch has it as well https://pytorch.org/docs/stable/generated/torch.transpose.html

_Originally posted by @a-corni in https://github.com/pasqal-io/Pulser/pull/764#discussion_r1954030546_

danielgaskins, 2026-09-20:
I’m happy to handle this one. Could you assign it to me? I’ll add `pulser.math.allclose()` with matching NumPy and PyTorch tolerance and `equal_nan` behavior, plus `AbstractArray.transpose(dim0, dim1)` backed by NumPy `swapaxes` or PyTorch `transpose`. I’ll cover NumPy and tensor inputs, tolerance handling, NaN comparisons, axis swapping, and gradient preservation.

HGSilveri, 2026-09-21:
> I’m happy to handle this one. Could you assign it to me? I’ll add `pulser.math.allclose()` with matching NumPy and PyTorch tolerance and `equal_nan` behavior, plus `AbstractArray.transpose(dim0, [...]

Sounds good @danielgaskins ! I have two additional requests:
- In `transpose()`, can you default `dim0=0` and `dim1=1` so that we do the usual transpose by default?
- Nice to have: use these new features where you see fit throughout the codebase (this doesn't have to be exhaustive)

closed 2026-10-02 by HGSilveri