pasqal-io/Pulser #819: Rework errors when a register doens't match a device

Yoric, 2025-03-04:
In https://github.com/pasqal-io/quantum-evolution-kernel, we're using `Sequence` to test whether a geometry can run on a given device. Essentially, that's

```py
        try:
            pl.Sequence(register=reg, device=self.device)
        except ValueError as e:
            raise CompilationError(f"The graph is not compatible with {self.device}: {e}")
```

This works as intended, but there are two issues:

1. `ValueError` is too generic. We could accidentally be catching unrelated errors. I'd prefer a custom `PulserError` or `RegisterError`, etc.
2. Pulser logs `ERROR:...` messages on the console/notebook, despite the fact that this is actually not an error. I'd prefer if these messages were part of the exception raised.

As far as I can tell, there is no other mechanism that will let me validate a register. Could we either rework the above two points or introduce another mechanism (e.g. `Sequence.check_register(register=reg, device=device)`) that I could use for this purpose, without the two aforementioned issues?

Yoric, 2025-03-13:
So, my plan is to introduce a hierarchy of errors and to build from there.

Note that there are currently 636 uses of `ValueError` in the repo, so we'll see how far I can get.

HGSilveri, 2025-03-13:
> So, my plan is to introduce a hierarchy of errors and to build from there.
> 
> Note that there are currently 636 uses of `ValueError` in the repo, so we'll see how far I can get.

I'll be the first to admit that we abusively use `ValueError` in some places but I'd ague that in most places it is being appropriately used (i.e. according to its [definition](https://docs.python.org/3/library/exceptions.html#ValueError)). 

If you are going for a big overhaul of the errors in Pulser, I have a few requests:
- Please be wary of backwards compatibility, we never know who is currently using a `try-except` block that expects this `ValueError`
- To the best of your ability, please try to split the work in small pieces, as that helps us digest it and review it
- Run what you have in mind by us first (especially if it's a lot of code) to avoid wasted effort

I'm confident you were already going to do all these things, it's me that would rather not leaving anything up to chance.

Yoric, 2025-03-13:
@HGSilveri I have a prototype, which returns a subclass of `ValueError`, so as not to break uses that catch `ValueError`.

I suppose that I should keep the existing error message strings, right?

HGSilveri, 2025-03-13:
> [@HGSilveri](https://github.com/HGSilveri) I have a prototype, which returns a subclass of `ValueError`, so as not to break uses that catch `ValueError`.
> 
> I suppose that we expect the existing error message strings to be part of the API, so we should keep them, right?

Sounds great! Yes, I'd say we want to keep the error message strings if possible.

commit 39387ca7 (2025-03-14): [DevEx] Replacing ValueError with more specific errors wherever possible - resolves #819

This should make it easier for client libraries to catch error we
raise and do something useful with them.
commit 695d7f3f (2025-03-19): [DevEx] Replacing ValueError with more specific errors wherever possible - resolves #819

This should make it easier for client libraries to catch error we
raise and do something useful with them.
commit 84579ad8 (2025-03-19): [DevEx] Replacing ValueError with more specific errors wherever possible (#828)

* [DevEx] Replacing ValueError with more specific errors wherever possible - resolves #819

This should make it easier for client libraries to catch error we
raise and do something useful with them.

* Applying feedback
closed 2025-03-19 by a-corni