pasqal-io/Pulser #824: De/Serialize EmulationConfig

sgrava, 2025-03-06:
Add methods to `EmulationConfig` to de/serialize itself.

### Description
Currently, pulser-cloud implements it own backend config (see here), which is difficult to maintain.
To avoid this dependency, `EmulationConfig` and `Results` should be able to de/serialize themselves and simply be passed back and forth through pasqal-cloud.

The solution is to have `form/to_abstract_repr ` to go from a string to a `EmulationConfig` and vice versa.
Internally, the implementation will be similar to what is already available for `pulser.Sequence`, i.e subclassing JSONDe/Encoder:
```python
class EmulationConfigDecoder(json.JSONDecoder):
    def default(self):
        ...

class EmulationConfigDecoder(json.JSONDecoder):
    def __init__(self, *args, **kwargs):
        ...

    def object_hook(self, obj):
        ...
        return obj
```
```python
class EmulationConfig:
    _decoder: Type[EmulationConfigDecoder] = EmulationConfigDecoder
    @classmethod
    def from_abstract_repr(cls, obj_str: str) -> EmulationConfig:
        return json.decode(obj_str, cls=cls._decoder)
```

### Acceptance criteria
- tested `form/to_abstract_repr ` methods in `EmulationConfig`

sgrava, 2025-03-06:
@HGSilveri @a-corni please, take a look if you have something you want to add to this issue.

a-corni, 2025-03-06:
Looks good to me :)

HGSilveri, 2025-03-07:
You have two `Decoder` in your example, I think you meant for the first to be an `Encder`. Otherwise, LGTM!

HGSilveri, 2025-03-25:
We still need the deserializer

PR #839 (2025-03-28): Add EmulationConfig deserializer
- [x] Add a deserializer for `EmulationConfig` and its subclasses
- [x] Unit tests 

Closes #824 .