pasqal-io/Pulser #831: Consider reworking the errors raised by (de)serialization

Yoric, 2025-03-17:
As a followup to #819 / #828 , it might be a good idea to consider also making the errors raised by (de)serialization a bit higher-level.

Yoric, 2025-03-24:
@a-corni So, what were your ideas there? As far as I understand, there are already exception specific to (de)serialization in https://github.com/pasqal-io/Pulser/blob/develop/pulser-core/pulser/json/exceptions.py .

Would you want to move them to `pulser.exceptions`? Make them subclasses of `PulserError`?

a-corni, 2025-03-24:
@Yoric Exactly, the question I raised during #828 was whether or not it would make sense to have them in the newly created https://github.com/pasqal-io/Pulser/blob/develop/pulser-core/pulser/exceptions.py, such that we define all the exceptions in one single file ? I think it would make sense to make them a subclass of PulserError

Yoric, 2025-03-24:
@a-corni Is this the kind of thing you have in mind? https://github.com/pasqal-io/Pulser/pull/835

If so, I'll continue the PR and add the rest of (de)serialization errors.

PR #835 (2025-03-26): [DevEx] Rework (de)serialization exceptions
Resolves #831. In #828, we reworked exceptions during the constructions of sequences. Let's now apply this design to (de)serialization.
closed 2025-03-26 by HGSilveri