pasqal-io/Pulser #850: Deprecate QutipBackend

pablolh, 2025-04-07:
QutipBackendV2 uses new generic API from `pulser.backend` and should be preferred over QutipBackend

pablolh, 2025-05-16:
@HGSilveri
@a-quelle mentions the documentation needs to be adjusted but we didn't talk about it, any opinion?

HGSilveri, 2025-05-16:
> [@a-quelle](https://github.com/a-quelle) mentions the documentation needs to be adjusted to use the new QutipBackendV2 (including all notebooks and results) but we didn't talk about it, any opinion?

That's right, we should replace every usage of `QutipBackend` in the docs by `QutipBackendV2`. I'm realizing now that the other backends on `pulser-pasqal` have not yet moved to the new interface, so this is potentially premature. Makes me wonder whether it is actually too soon to deprecate `QutipBackend` after all

PR #907 (2025-07-11): Deprecate QutipBackend
- Deprecates `QutipBackend`
- Replaces it with `QutipBackendV2` on all tutorials **except for the Backend Execution page**, which will be tackled in #875 

Closes #850 .
closed 2025-07-11 by HGSilveri