pasqal-io/Pulser #852: Clarify `No DMM dmm_0 in the device` error message

Yoric, 2025-04-11:
When reading this error message, it's not clear to me whether:

a. There's no `dmm_0` because the device does not support DMM channels.
b. There's no `dmm_0` because of a typo.
c. There's no `emm_0` because I forgot to declare it somewhere.

Yoric, 2025-04-11:
Could we possibly tweak the message to the following? This would remove any ambiguity.

`No DMM dmm_0 in the device: this device supports DMMs [ ... ]` / `No DMM dmm_0 in the device: this device does not support DMM`

a-corni, 2025-04-11:
Yes, I would be in favour of using your first option. Combining your proposition and @awennersteen's:

f"No DMM called {dmm} is available in the device. The device supports the following DMM channels: {device.dmm_channels}"

That makes a good first issue !

closed 2025-05-14 by a-corni