pasqal-io/Pulser #887: Replace tutorials on simulations with noise

HGSilveri, 2025-06-17:
The tutorials on simulation with noise are enforcing outdated workflows (e.g. usage of `QutipEmulator` and `SimConfig`), are cumbersome to maintain and don't convey the information efficiently.

We should replace them with a page containing:
- An explanation of the `NoiseModel` class
- A brief explanation of each type of noise and its associated parameters
- How to include noise in the emulations via the backends

a-corni, 2025-06-19:
About the first two points, is it okay to display the NoiseModel class directly https://pulser.readthedocs.io/en/stable/apidoc/_autosummary/pulser.noise_model.NoiseModel.html#noisemodel ? As the device in hardware https://pulser.readthedocs.io/en/stable/hardware.html#Reading-through-the-Device's-specifications

HGSilveri, 2025-06-19:
I think so! Potentially with some cleaning up of the docstring

closed 2025-07-01 by a-corni