pasqal-io/Pulser #928: Strange results for atomic motion noise

HaroldErbin, 2025-09-13:
I have tried the last dev version with the atomic motion noise (#881, #915). Simulations with register noise give very different results compared with the simulations I made with position noise implemented by hand (= averaging several simulations with register differing by noise on atom positions).

I computed $U$ and $w$ from the values of $\sigma^{xy}_r$ and $\sigma^z_r$ using the formulas above, and checked that `_register_sigma_xy_z` gives back the original values, so I don't think it's a problem from the parameters (though it would have helped to know the parameters directly). Since this is an ongoing work, I cannot be more specific.

Moreover, I am surprised that the simulation does not give a confidence interval (like with SPAM) since the noise is stochastic.

HaroldErbin, 2025-09-19:
I do know that Lindblad is exact. My ticket is about the new register noise from #915, and I thought that it was not described using Lindblad but a loop, in which case I would expect to get a list of states and be able to compute average and std like I do for SPAM. But no std is shown for my graphs. I did not look into the intermediate steps as I don't have time for this now, I am just warning that the display is not what I expect for register noise (knowing that my pipeline works for other types of noise, be it Lindblad, SPAM, laser, etc.). As I said in my previous message, I think that you gave the explanation by saying that register is not updated between steps in some cases as it should.

Here is a snapshot of my simulation with various sources of noise (reg. noise = code from #915, pos. noise = manual loop over different registers, complete model = dissipation, SPAM, laser, Doppler, pos. noise).

[image]

[image]

As you can see, the two computations with reg. noise are just off (and pure reg. noise has no variance). I would have expected reg. noise to be inside the area of pos. noise.

HaroldErbin, 2025-09-24:
I was aware of #934 but I don't think that the Doppler noise explains the difference between the two curves with "reg. noise" and the rest, since the complete noise model (in cyan) does include Doppler noise, and its general trend is not much different from the other simulations. In any case, I will try the new feature when I find time.

a-quelle-pasqal, 2025-09-26:
See https://github.com/pasqal-io/Pulser/pull/942

HaroldErbin, 2025-10-01:
I tried out the last version, and I think that it fixes both problems. First, standard deviation is correctly obtained. Second, the two curves are now much more similar, and the differences could be due to random fluctuations.

[image]

closed 2025-10-01 by HaroldErbin